agimus-project/agimus-controller #104: Setup The Warm Start Factory

MaximilienNaveau, 2024-12-05:
I am wondering if the information necessary for the setup and generate of the WarmStart class should be contained in the OCP class or in class specific parameters.
For the moment we have 3 warm start implementation in mind:
- "from_gravity_compensation" -> `setup` needs the robot model, `generate` the current q
- "from_previous_solution" -> `setup` maybe gets the OCP discretization dt and the MPC frequency, `generate` needs the previous solution  
- "from_diffusion_model" -> `setup` might need the generative model, `generate` might need a description of the environment/obstacles

_Originally posted by @MedericFourmy in https://github.com/agimus-project/agimus_controller/pull/97#discussion_r1869039586_

TheoMF, 2024-12-06:
For agimus_controller at the first solve we use inverse dynamics as warm start, same as the references used in the costs, so the `generate` function of the WarmStart class would need the control plan made by the ocp

MedericFourmy, 2024-12-06:
Sorry, I'm not sure to understand your point. You mean that in the current implementation you use `tau = pin.rnea(q,dq,ddq)` for each point of the trajectory as a warm start right?

>  control plan made by the ocp
You mean made by the planner?

TheoMF, 2024-12-06:
> Sorry, I'm not sure to understand your point. You mean that in the current implementation you use `tau = pin.rnea(q,dq,ddq)` for each point of the trajectory as a warm start right?

well the point is we were talking about the warm start we would use and we were missing one, yes in the current implementation I'm using the formula you are mentioning

> > control plan made by the ocp
> 
> You mean made by the planner?

well the computation of the control is made in the ocp (maybe it will be done elsewhere after the refactor) but it's based on informations from the planner