agimus-project/agimus-controller #109: I am wondering if the information necessary for the setup and generate of the WarmStart class should be contained in the OCP class or in class specific parameters.

MaximilienNaveau, 2024-12-09:
I am wondering if the information necessary for the setup and generate of the WarmStart class should be contained in the OCP class or in class specific parameters.
For the moment we have 3 warm start implementation in mind:
- "from_gravity_compensation" -> `setup` needs the robot model, `generate` the current q
- "from_previous_solution" -> `setup` maybe gets the OCP discretization dt and the MPC frequency, `generate` needs the previous solution  
- "from_diffusion_model" -> `setup` might need the generative model, `generate` might need a description of the environment/obstacles

_Originally posted by @MedericFourmy in https://github.com/agimus-project/agimus_controller/pull/97#discussion_r1869039586_

MedericFourmy, 2024-12-09:
Isn't this a duplicate of https://github.com/agimus-project/agimus_controller/issues/104?

MaximilienNaveau, 2024-12-09:
We can keep the discussion around #104 indeed.

closed 2024-12-09 by MaximilienNaveau