agimus-project/agimus-controller #138: Enable multithreading for OCP + Modify compilation option for the robot

jmirabel, 2025-01-20:
Multithreading is implemented using OpenMP. We want to enable OpenMP only inside crocoddyl. This is done using a CMake flag inside crocoddyl.

Tasks to make it work:
- [ ] based on the *development* Dockerfile, make an *experiment* Dockerfile that only differs on those points.
  - compiles all code with `-03` option and without debug symbols
  - compiles crocoddyl with OpenMP support
- [ ] benchmark the performances of crocoddyl with OpenMP on miyanoura to assess that we are indeed able to have better performances. The test can first be ran in C++. However, it should be tested with Python.
- [ ] Make sure that future experiments uses the *experiment* image generated using the *experiment* Dockerfile.

Kotochleb, 2025-01-24:
Docker is build now with `-O3` and OpenMP enabled in crocoddyl. It is still on `RelWithDebInfo` instead `Release`

jmirabel, 2025-01-29:
> It is still on `RelWithDebInfo` instead `Release`

Why so?

Are you sure your `-O3` is not overwritten by `RelWithDebInfo`, which contains `-O2` ?

jmirabel, 2025-02-20:
Can be closed as soon as https://gitlab.laas.fr/agimus-project/agimus_dev_container/-/merge_requests/21 is merged.

closed 2025-02-28 by jmirabel