agimus-project/agimus-controller #143: [MPC] create trejectory buffer logic

MaximilienNaveau, 2025-01-21:


jmirabel, 2025-01-24:
We haven't written down the requirements from our discussion this week.

I remember that you said we would specify the times as
```yaml
T: 42
dt: 0.01
dt_factors: [ 1, 1, .... 2, 2, ... ]
```

I have played on my side with increasing time step in OCP. Specifying the time steps using `dt` and `dt_factors` is fairly tedious and very error-prone. I personally used:
```yaml
T: 42
dt: 0.01 # Minimum time step, which has to be the first time step.
dt_n_seq:
  - [ 0.01, 18 ] # 18 slot of 0.01 seconds
  - [ 0.02, 14 ] # 14 slot of 0.02 seconds
  - [ 0.05, 10 ] # 10 slot of 0.05 seconds
```
When loading, I require that:
```python
timesteps = sum(([sdt,] * sn  for sdt, sn in dt_n_seq]), [])
# check first time step
assert timesteps[0] == dt
# check number of time steps
assert len(timesteps) == T
# increasing time steps
assert all(t0 <= t1 for t0, t1 in zip(timesteps[:-1], timesteps[1:]))

MaximilienNaveau, 2025-01-24:
The orginal one where indeed very tedious to parse.
I like the second approach but can't we keep the best of the 2?

```python
T: 42
dt: 0.01 # Minimum time step, which has to be the first time step.
dt_n_seq:
  - [ 1, 18 ] # 18 slot of 1 * dt=0.01 seconds
  - [ 2, 14 ] # 14 slot of 2 * dt=0.02 seconds
  - [ 5, 10 ] # 10 slot of 5 * dt=0.05 seconds
```
And then:
```python
timesteps = sum(([factor * dt,] * sn  for factor, sn in dt_n_seq]), [])
# check first time step
assert timesteps[0] == dt
# check number of time steps
assert len(timesteps) == T
# increasing time steps
assert all(t0 <= t1 for t0, t1 in zip(timesteps[:-1], timesteps[1:]))
```

jmirabel, 2025-01-24:
>  I like the second approach but can't we keep the best of the 2?

I didn't get your question. If you also prefer to specify the time steps as
```yaml
T: 42
dt: 0.01 # Minimum time step, which has to be the first time step.
dt_n_seq:
  - [ 1, 18 ] # 18 slot of 1 * dt=0.01 seconds
  - [ 2, 14 ] # 14 slot of 2 * dt=0.02 seconds
  - [ 5, 10 ] # 10 slot of 5 * dt=0.05 seconds
```

then let's go for this. I don't mean to have this alongside the `dt_factors`. I mean to have this **instead of** the `dt_factors`.

MaximilienNaveau, 2025-01-24:
Yes of course I understand it's instead.
What I meant is if you keep the factor as integer you can ensure the dts are actual multiple of dt.
It feel less error prone...
Anyway if you think my last message is ok

jmirabel, 2025-01-24:
Ah, I understand what you meant. I copy-pasted without noticing the change from 0.01 to 1...

I don't think using time would be very error prone for most situations where you only have a few different times. I guess that will be our case though I'm not sure.

Anyway, I'm fine with using integer factor rather than actual time. However, can you rename from `dt_n_seq` to `dtfactor_n_seq` please ?

MaximilienNaveau, 2025-01-27:
> Anyway, I'm fine with using integer factor rather than actual time. However, can you rename from `dt_n_seq` to `dtfactor_n_seq` please ?

Sure thing!

jmirabel, 2025-02-20:
I think this can be closed, right ?

MaximilienNaveau, 2025-02-20:
Yes let's close this one

closed 2025-02-20 by MaximilienNaveau