agimus-project/agimus-controller #153: Warm start classes

jmirabel, 2025-01-24:
I have two remarks regarding the warm start classes.

- [ ] `WarmStartReference.generate` calls `rnea` for each point. We have a performance loss here. We only need to use RNEA algo once for each point regardless of how many times they are used by successive `generate` calls.
- [ ] `WarmStartBase.generate` should provide as a default behavior a warm start based on the previous solution.

The second point has a rather high priority to me as it is what I think we need for MPC with collision avoidance.

jmirabel, 2025-01-24:
FYI @kzorina @MaximilienNaveau

MaximilienNaveau, 2025-01-24:
The second point I disagree. We should provide such a warmstart but not as base. As an educated guess from the one picking the warmstart they want.
Hence I would suggest to implement a daughter class that implements such warmstart.

jmirabel, 2025-01-27:
I have played with the second point using a child class of WarmStartBase. This is implemented in #124

MaximilienNaveau, 2025-07-08:
Has anyone investigated the issue with the RNEA called at every iteration?
@TheoMF @kzorina @Kotochleb @jmirabel

closed 2026-10-12 by MaximilienNaveau