agimus-project/agimus-controller #190: Make LFC run with real time permissions

jmirabel, 2025-02-20:
At the moment, LFC is ran with normal permissions which can causes issues.

jmirabel, 2025-02-20:
@MedericFourmy If you don't feel like doing it, I think you can leave this to @Kotochleb next week.

jmirabel, 2025-02-20:
Are you sure that, inside the docker, ros2_control can enforce the real time capabilities?
I have had previous experience on this. I needed to pass custom flags when creating the docker container. Otherwise, it just did not work.

It was something like `--cap-add SYS_NICE`.

@nim65s might be of some help on this as well.

MaximilienNaveau, 2025-02-20:
I would try the classic `nice(-20) ` inside the ros2 node as a first try.
Then we can start from that if this is not enough

jmirabel, 2025-02-20:
> I would try the classic `nice(-20) ` inside the ros2 node as a first try. Then we can start from that if this is not enough

If you do that, you should check in `htop` on the host whether your process has the priority you manually set.

kzorina, 2025-02-20:
we actually tried `nice` with Vladimir, but there was no effect on the host machine in `htop`
and if we run `sudo nice`, it was crashing

jmirabel, 2025-02-20:
I expect that you must create your docker container with the sys_nice capability.

Kotochleb, 2025-02-24:
@jmirabel docker by default will not allow realtime, but I didn't know about the priorities. The dev containers we use have `--privileged` and this should be equivalent to `--cap-add SYS_NICE`, meaning this should not be a problem.

> we actually tried `nice` with Vladimir, but there was no effect on the host machine in `htop`
> and if we run `sudo nice`, it was crashing

@kzorina when running it with `sudo` you have different environment variables than as a user. You can do:
```bash
sudo su
source ~/ros2_ws/install/setup.bash
# whatever you were running
```
Other option is to run the LFC, then grab PID of it and change manually it's niceness. This might be a simpler option for the moment. 

---
Also for true real time here is [documentation](https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html#determinism) from ROS control on the RT priority. For the docker in compose this is required:
```yaml
    ulimits:
      rtprio:
        soft: 99
        hard: 99
      memlock:
        soft: 102400
        hard: 102400
```
Tho it would require some checking to see how to use it in dev container. We could consider moving the LFC to another computer if needed. There is one PC with RT kernel at LAAS next to panda.

MedericFourmy, 2025-02-25:
We found the following solution to work:
- switch to the admin user (`sudo su`) before running `ros2 launch agimus_demo_05_pick_and_place bringup.launch.py`
- add `nice -n -20` to the controller manager node prefix

Kotochleb, 2025-02-26:
I added an option when launching panda to set process affinity in this PR https://github.com/agimus-project/agimus-demos/pull/51. After testing it, it looked like all threads were pinned to a single core

MaximilienNaveau, 2025-02-26:
Maybe you can add at least one more core to there so rclcpp has a bit more flexibility?
Because 1 core is roughly 2 threads no?
So 4 threads in parallel is nice for the callbacks.

jmirabel, 2025-02-27:
> We found the following solution to work:
> 
>     * switch to the admin user (`sudo su`) before running `ros2 launch agimus_demo_05_pick_and_place bringup.launch.py`
> 
>     * add `nice -n -20` to the controller manager node prefix

Out of curiosity, have you tried to follow [those instructions](https://control.ros.org/master/doc/ros2_control/controller_manager/doc/userdoc.html#determinism)?

Although it works, using `sudo` isn't the best method.

jmirabel, 2025-02-28:
After some internal discussions, we will move to using a real time kernel both in Prague and in Toulouse.

Kotochleb, 2025-03-07:
Good new is that the approach with docker works-ish. It still need source build of pinocchio as the one installed from apt doesn't work (yay...). After testing real time priorities seems to be applied correctly, as controller manager has priority -51 in htop. I hope this, and figuring out how to handle launching LFC parameters will be all that is left

Kotochleb, 2025-03-14:
Both PRs got merged and were tested both at LAAS (by me) and at CTU independently by @MedericFourmy. Everything should work just fine

closed 2025-03-14 by Kotochleb